tagyoureit/nodejs-Pentair #2: A tiny bit of nearly useless info

rflemming, 2016-06-09:
I was just tinkering with stuff and found one tiny bit of info related to one of the currently unused packet fields.  The 10th field at a minimum contains info about the units being used.  0 = English, 4 = Metric as seen below:

15, 16,  2, 29,  6, 40,  0,  0,  0,  0,  0,  0,  0,  0,  3,  0, 32,  4, 81, 81,  0,  0, 63, 59,  0,  0,  3,  0,  0,181,198,  1, 13,  3,234
15, 16,  2, 29,  6, 40,  0,  0,  0,  0,  0,  0,  0,  4,  3,  0, 32,  4, 27, 27,  0,  0, 17, 15,  0,  0,  3,  0,  0,132,225,  1, 13,  3,18

You'll notice the temperature values all change.  It's also likely that this stores some other info just like the heat mode since 0 and 4 don't really make any sense otherwise.  I tried a few other random settings, but couldn't find anything that also toggled that value.

tagyoureit, 2016-06-09:
That's awesome.  I am on a business trip now so might take me a couple of weeks to add this, but I'll certainly do it.  Check out the wiki to see what I've learned about the config settings.  You might be able to help out there, as well.

tagyoureit, 2016-06-22:
Added in 0.0.4.  Thx.

closed 2016-06-22 by tagyoureit

rflemming, 2016-07-04:
Just a quick followup to this.  Like I suspected more info is store in this particular byte.  I don't have it all worked out, but here's what I do know about a couple of the bits:

1 - Auto, Service/Timeout
2 - 
4 - English, Metric
8 - 
16 - 
32 - 
64 - 
128 - Timeout?
256 - 

For example:

Mode = Auto, Units = English => 0
Mode = Service, Units = English => 1
Mode = Timeout, Units = English => 129 (Not exactly sure of the relationship with the first bit)
Mode = Auto, Units = Metric => 4
Mode = Service, Units = Metric => 5
Mode = Timeout, Units = Metric => 133

Obviously there's more to this.  At a minimum, it's safe to say everything mentioned here:

http://www.autelis.com/wiki/index.php?title=Pool_Control_(PI)_HTTP_Command_Reference

is in there somewhere.  So my guess is perhaps these other bits contain the freeze protection state, as well as the state for the various sensors.

rflemming, 2016-07-04:
The byte after the previously discussed one appears to change when the pool goes into solar heating mode:  It goes from 3 to 51 when the solar becomes active, or in the case below the opposite (the solar turned off):

[165, 10, 15, 16, 2, 29, 13, 59, 32, 0, 0, 0, 0, 0, 0, 0, 51, 0, 32, 4, 87, 87, 32, 0, 82, 108, 0, 0, 3, 0, 0, 98, 183, 1, 13, 4, 98]
[165, 10, 15, 16, 2, 29, 13, 59, 32, 0, 0, 0, 0, 0, 0, 0, 3, 0, 32, 4, 88, 88, 32, 0, 82, 107, 0, 0, 3, 0, 0, 98, 183, 1, 13, 4, 51]

Not sure if this might be different when using an actual heater, but the odds are likely.  I don't have a heater though to confirm.

rflemming, 2016-07-05:
I also recently found this:

https://carlstrom.com/pool/compool/
https://carlstrom.com/pool/compool/protocol.txt

I think there's some helpful info there, but it does look like the protocol may be slightly different because the order of some of the bytes doesn't match up with what I'm seeing.

tagyoureit, 2016-07-18:
>Just a quick followup to this. Like I suspected more info is store in this particular byte. I don't have it all worked out, but here's what I do know about a couple of the bits:
> 1 - Auto, Service/Timeout

Service and Timeout are two different states... I'm surprised the unit of measure is lumped in the bit.  Then again, not much is surprising me at all anymore.  Good find.  On my to-do list to test on my system.

> I also recently found this:
> https://carlstrom.com/pool/compool/

Yes, I've seen this as well but it is worth going over again.  I believe this is for the previous(?) generation of Pentair controllers, but some items like the delays might still be valid.