tagyoureit/nodejs-Pentair #5: Trouble reading data

leftyfl1p, 2016-09-05:
I tried setting this up but I think that the server isn't reading the values correctly. I'm using the  EZSync you have linked in the readme (before I read about the $5 one :/ ) and running on a raspberry pi. 

The pi only has 1 usb device plugged in so the adapter is at ttyUSB0. 
Here's output of when trying to run the server and ticking one of the boxes on the main web page: http://pastebin.com/DkGyEVv1

Here's a huge data sample from the device with `od -x < /dev/ttyUSB0`: http://pastebin.com/GvADLwqU (which has lots of repetitions of `ff a5ff` which leads me to believe everything is hooked up correctly). If I toggle something manually with the wireless remote I see activity in the terminal as well.

What problems were you having with the EZSync? Hope that this isn't related.

tagyoureit, 2016-09-05:
Hi,  
I never had trouble reading with the EZSync adapter.  The trouble I had was writing back to the bus.  Not sure why... but I tried all combinations of the Data+, Data- and ground.  I never hooked up the +5V because I'm not an electrical engineer and didn't want to short anything out.  Since the other $5 adapter doesn't have a ground or +5V I don't imagine this is the problem.

Your output looks fine in the od -x command.  The app should be able to read your output.

Questions for you:
1) What system controller do you have (eg Intellitouch?)
2) Is there any output at all in the console?   What is happening that is giving you trouble?

You could try two things:
a) Comment out the lines that write back to the bus (Lines [1602](https://github.com/tagyoureit/nodejs-Pentair/blob/master/index.js#L1602)-1604)
b) Try setting the log level to Debug (or Silly) at line [37](https://github.com/tagyoureit/nodejs-Pentair/blob/master/index.js#L37).  This will give you any/all debug & error messages.

tag

leftyfl1p, 2016-09-05:
Hello,

I only have Data + and - wired.

The panel says easy touch. I'm not 100% sure on all the different names all the pentair systems have. Here's what it looks like: http://imgur.com/a/WFA3l .

The first pastebin was all the output that I got from running it normally. I turned the log level to silly and this is all I'm getting: http://pastebin.com/yReMpmy2

Nothing else gets logged even if I start changing things on the wireless remote.

The trouble is that I was at least expecting the web UI to provide info like the System Config Status or any info on the pumps, as they all say undefined.

Thanks

tagyoureit, 2016-09-05:
Do you have the od -x running at the same time as this app (or any other apps listening to the /dev/ttyUSB0 channel)?  I think the od -x will prevent Node (or SerialPort) from opening the port.  

I'll have to think about this a little bit.  I could add some debug code to see if the SerialPort is able to correctly open the port as I don't have any debug messages in there regarding that.  If you have the right port, which you do, the app should be able to read the port as well... hmmm.

tag

leftyfl1p, 2016-09-05:
I did actually have od -x running at the same time. Rebooted and tried again but same result. I also adjusted the temp a couple of times and got some info logged back but the web UI didn't change. Here's the output http://pastebin.com/LdL6ygdq

tagyoureit, 2016-09-06:
> adjusted the temp a couple of times and got some info logged back

Can you expand on this?  I didn't see anything new in the last pastebin.  What was logged back that was new?

leftyfl1p, 2016-09-06:
I adjusted the temperature of the pool via the easy touch panel manually. There was likely nothing new in the latest pastebin, I sent it in case there was something different in the messages in the logs (I haven't read up on the wiki yet to know what they mean).

tagyoureit, 2016-09-06:
You need to adjust the pool temp via the ScreenLogic.  If you adjust the temp on the Easy Touch, it won't trigger the Screenlogic to request all the configuration settings again.  Log in via your phone or computer app and then adjust the screen.  

If you can change the temp through either of those interfaces, unplug the Screenlogic module for at least 10 seconds and hopefully, it will request all the settings again.

However, you should still be getting _some_ messages through the app.  The fact that nothing is coming through means something else has the port locked open, or it isn't the right port.

Try 'ls /dev/ttyUSB0' from the terminal prompt to make sure the RasPi can see the USB RS485 module.

tagyoureit, 2016-09-06:
Could you also try saving the following to a new read_only.js file and running it?  It should be similar to od -x output.  If not, then Node/SerialPort isn't able to read from your USB device.

```
'use strict';
var serialport = require('serialport');

var port = new serialport.SerialPort("/dev/ttyUSB0", {
    baudrate: 9600,
    databits: 8,
    parity: 'none',
    stopBits: 1,
    flowControl: false,
    parser: serialport.parsers.raw
});

port.on('data', function (data) {
    console.log('Received: \t', data);
});
```

leftyfl1p, 2016-09-07:
I don't actually have a ScreenLogic. I was going to use this as an alternative. Unfortunately the pool is at a vacation home and the Pi was being powered from an outlet that's exposed to the elements so I couldn't leave it there when I left, so I can't even remote in at this time. I'll probably be returning this weekend so I'll get you that info then.

leftyfl1p, 2016-09-10:
Here's the output when doing
```
 $ ls /dev/ttyUSB0
/dev/ttyUSB0
```
which I'm assuming is what should show up.

It also looks like the data isn't being read correctly after trying the test program. http://pastebin.com/GfAaqYtZ
I suppose I'll have to try out the $5 module. Even though I'm certian I wired everything correctly, is it possible this could happen from incorrect wiring?

tagyoureit, 2016-09-10:
Your output actually looks fine. The ff 00 ff a5 is what you want to see. 

Can you change the following debug parameters in the app?

Var logLevel=1;
Var logPumpMessages=1;

Also, comment out all lines in getConfiguration so the code doesn't try to write at this time.

leftyfl1p, 2016-09-10:
After those changes I now at least have some information coming through: http://pastebin.com/yKNHbFBg

http://imgur.com/a/1e9JM schedules tab is blank.

tagyoureit, 2016-09-11:
This is where writing back will help you. The getConfiguration requests the schedules along with circuit names etc. 

Do you have any other wired controllers?  If so, you should try starting the app and then disconnecting and reconnecting any other controllers. This may get them to request the configuration information that will then be displayed in the app. 

And, yes, spend the $5 for the other adapter. :-)

tagyoureit, 2016-09-20:
Lefty,
In 0.1.0, I now have flags you can set for Intellitouch, IntelliCom (limited), or pumps only.  You should try this version and set the flags as appropriate.  I have included quite a bit more pump info now.
tag

closed 2016-09-30 by tagyoureit

leftyfl1p, 2016-11-05:
For anyone that reads this in the future: I actually just tried the $5 adapter and it worked fine.