tagyoureit/nodejs-Pentair #12: Network Access

arrmo, 2016-09-30:
Hi,

Perhaps a dumb question, but ... any thoughts how to change this to read / write to a network port, rather than serial directly? I'd like to use socat (or nc), to run this on another machine on the network - to be able to put a low end controller (e.g. Rasperry Pi Zero) out by the pool controller), but have "another" machine handle the heavy lifting.

Yes, I have socat working - so just need this to talk to a network port (socket), not the serial port directly. The data seen on the network port is identical to the serial port.

Thanks!

tagyoureit, 2016-09-30:
Hi Armo,

I'm not familiar with socat, but I already have [Socket.io](http://socket.io/) built-in (look at my web pages in the app) and it sounds like that is similar to socat(?).  It wouldn't transfer the serial port data but your other computer could read in the info and just be a client to this app.

Are you sure the RasPi Zero can't run this application?  If it can (and I don't see why not), then you might be trying to solve a problem that you don't need to solve.

tag

closed 2016-09-30 by tagyoureit

arrmo, 2016-10-01:
Will dig into it a bit more, provide an update back to make this more general (if I can!).

Now, to figure out how to get this working - I admit, a big confused by that ... :(.

Thanks!