tagyoureit/nodejs-Pentair #14: NetBeans Debug

arrmo, 2016-10-02:
Hi,

My apologies in advance - I know this is me, but not quite sure how to get this up and running (for debug). I would like to help out, but stuck early on ... :(.

I have my interface all up and working, have been playing with that an Python for a little bit - so I know the HW and connection are all working. But coming across this - I think it's great, want to get involved ... but not quite sure how to get this up and running (i.e. stuck on the JS part of it).

Any info you can provide would be much appreciated! I have the files all local now (copied for now), and grabbed socket.io ... but getting a bunch of errors (like below) when trying to execute index.js in NetBeans,
Failed to load resource: net::ERR_EMPTY_RESPONSE (06:23:40:335 | error, network)
  at http://localhost:8383/socket.io/socket.io.js
Failed to load resource: net::ERR_EMPTY_RESPONSE (06:23:40:337 | error, network)
  at http://localhost:8383/main.js
Failed to load resource: net::ERR_EMPTY_RESPONSE (06:23:40:581 | error, network)
  at http://localhost:8383/favicon.ico

Do I need to set all this up in an Apache Server? Would like to debug locally for now, but perhaps that's not possible.

Again, sorry for the dumb questions!

tagyoureit, 2016-10-02:
Hi Armo,

I would try to get this up and running with NodeJS (native) first, and then try to use Netbeans after that as a next step.  

Basic steps to get you this far:
1. Install Nodejs.  (https://nodejs.org/en/download/)
2. Update NPM (https://docs.npmjs.com/getting-started/installing-node).  
3. In the root folder of nodejs-Pentair (where package.json exists), run 'npm install'  This will automatically install all the dependencies (serial-port, express, sockets.io, etc).
4. Run the app by calling 'node index.js' (again, in the root directory).  It should now run properly.

You don't need to download/install socket.io or others separately.  

> Do I need to set all this up in an Apache Server?

No.  Express (included as a dependency and installed via NPM - https://www.npmjs.com/package/express) _is_ a web server.  When you start up 'node index.js' it will automatically load/launch Express, Socket.io, and everything else you need to get running.  You can access the web pages through http://servername:3000/index.html.  Port 3000 is where express is configured by default (and in the app).

If you get all of this working, then try Netbeans.  I haven't used it myself so can't talk about that specific process.  Let me know how it goes!

tag

arrmo, 2016-10-02:
Thanks for the detailed comments - will give this a try, get back to you. If it works ... perhaps add this to the readme? Just a thought, your call of course!

I have Apache set up and running already, for other pages. Would like to get this integrated there as well, but let's try that second ... ;-).

Thanks again!

arrmo, 2016-10-02:
OK, it's working - thanks! I want to get it running remotely (much faster than on RPi) - to debug that I can use NetBeans ... Node.js support is built in now (more info here, https://netbeans.org/community/releases/81/).

I admit, new to Node.js, need to figure that one out ... ;-). But also, wanting to get bootstrap working - do you want any updates proposed back to you (fork, and create pull requests)?

Thanks again!

tagyoureit, 2016-10-02:
I'm happy to have any and all help on the project. But can you explain what you mean by adding a bootstrap to the application?  There are many different ideas of bootstrapping out there so if you could explain what you want to accomplish (and how) that would be helpful.

arrmo, 2016-10-02:
Yes, sorry! I mean this ... http://getbootstrap.com/

Makes the html very nice on mobiles also - this is what I'm after. As for network access ... want to be able to run it on the (slow) local machine (which may just be a router!), or remote faster machine.

Thanks!

tagyoureit, 2016-10-02:
Well, being that my HTML is admittedly horrible, yes please do feel free to implement Bootstrap. There are some articles I see that show you can use Bootstap with express. Let's break this into three parts:
1) bootstrap on express
2) bootstrap with apache for your web server
3) your network access approach

Is it fair to say we are looking at these three separate items?  Or do you see it differently?

arrmo, 2016-10-02:
Hi,

Well, my HTML may only be marginally better ... LOL. But no worries, we'll work through it. I see it very similar, but slightly different order (below) ... but please do comment!
1) Network access - I need to take care of this, as debugging on RPi is VERY slow. This is to make the rest go quicker.
2) Bootstrap on Express - as you say, let's get this working on Express
3) Apache ... but this is a distant third. If Express works, I'm not super worried about this one.

Thoughts?

Thanks!

tagyoureit, 2016-10-02:
Sounds good to me.  For the network access, it would be possible, but honestly might be a chunk of work:
[Pool] <-- RS485 --> [RasPi Serialport --> Socket.io client] <-- Ethernet --> [Socket.io -->nodejs-Pentair]

The [RasPi-->Socket.io client] would just need to be a small app that is a pass through (RS485->Ethernet and vice versa).  In the nodejs-Pentair you would need to replace both serialport read and write with Socket.io functions.  

The [Raspi-->Socket.io client] doesn't exist now.  You would need to build that from scratch.

In the nodejs-Pentair app there would only be (theoretically) 2 places you need to change the code: in the sp.on(data) where it is read and the write function sp.write().

Maybe doable, but not sure how it could be integrated for the majority of users who want to plug into RS485 directly.  Maybe another configuration option if you want them to co-exist at the same time?

And just for curiosity, once you are "done" with debugging, would you then rely solely on the RasPi Zero to run the program or would you keep this configuration?

The rest sounds good...

arrmo, 2016-10-03:
The network access isn't really all that bad. What you have above is correct, but not as bad as it seems. I have an RS485-USB adapter plugged in to the RPi USB port ... then just use socat (https://linux.die.net/man/1/socat, single line running with daemon => service). It actually allows multiple simultaneous connections even. So no code to write there (OK, just the service script) ... then use net in nodejs-Pentair.

Actually, I already have it working! Check out the commit I just submitted. And yep, reading your note above - that's what I changed. I admit, I struggle with callbacks, and the logic for the parameters ... so on the network write, no error check right now. That can come later.

If you want to try it out - I can create a pull request. There is a variable that can be set, so then it's serial (like before) ... or if it's set for network, then the same code will work.

In the end - not sure. I likely will actually use a router (running Tomato firmware) ... free up the RPi again.

Thoughts?

tagyoureit, 2016-10-03:
Wow!  That's pretty slick.  I don't think you changed a lot, but Github sure makes it look like it!   [Compare](https://github.com/tagyoureit/nodejs-Pentair/compare/master...arrmo:master)

You should create a pull request.  I'm happy to include anything that helps move the project forward.  And I'll trade you... I'll add the NodeJS instructions that I provided you if you also add some documentation around the socat/net commands and usage (include remote setup, why/when it is useful, etc!)  I might even use it myself because I could really see the value of debugging locally.  (Did you get that working yet?)

If you want to join the conversation, we have a small (but active) group of us chatting on [Gitter](https://gitter.im/pentair_pool/Lobby).  Much easier (and realtime) compared to this Issues section.

closed 2016-10-03 by tagyoureit

arrmo, 2016-10-03:
I agree - I really didn't change all that much ... heck, it only took an hour or two, I couldn't have changed that much ... ;-).

Will create a pull request - NP. And the trade sounds good. I can provide the daemon / service file itself, and explain it ... you bet. Perhaps add a directory for scripts (name suggestion?), and I can store the file I use in there? Just let me know.

On the debugging - yep, it's working great ... using Netbeans, so then I can step through the code, full IDE debugger. Very nice!

Gitter ... OK, that one is new to me. What is it?

Thanks!

tagyoureit, 2016-10-03:
Gitter is a free, realtime chat app (very similar to Slack).  Good for developers/hackers to chat, share code and build a community.

This may seem nitpicky, but can you update your new files so all the extraneous changes don't appear?  I think it is most likely just due to formatting of Netbeans or whatever.  Normally, I'd just accept it, but the reason that it's important this time is that I have some changes off-line I'm going to commit in the next ~24 hours and some of the code it thinks you did change (but really didn't) are actually changed in my version.  I don't want it to go all haywire when I try to merge your changes with my next update.  👍   You should just be able to copy/paste everything else from the original back into your modifications and call it a day.

arrmo, 2016-10-03:
Hi,

On the changes - I can try to do that ... or if you push your changes, I can merge them in to my files, then update the Pull Request. Thoughts?

Will get set up on Gitter, easier to chat there it sounds like?

Thanks!

tagyoureit, 2016-10-03:
That sounds good. If I push up my changes you can merge, or clean up what
you have. I want you to get credit for the improvements.

arrmo, 2016-10-03:
Sure - just let me know when you push your updates, and I'll merge them in. Once you get to know me a bit better you'll realize - I don't care about credit, let's just get it all working together ... :-).

BTW, I'm Russell also - that's way too funny!

arrmo, 2016-10-04:
Not sure how else to ask this question - so adding it here (but yell if there is a better way). I have a very rough starting point file, just to show what may be possible with Bootstrap (only partial, no JS usage yet ... don't get too excited ... :-)). But if you want to look at it - how to get it to you?

Thanks,
... Russell

tagyoureit, 2016-10-04:
You can share files in the Gitter site. Join the right room and just
click/drag the file(s) into the chat window.

tagyoureit, 2016-10-04:
BTW, yes, nice to meet another Russell.  Thanks for helping out!  And now I know that the footer from my phone gets appended when I reply via email, LOL.  😄