tagyoureit/nodejs-Pentair #24: Provide external http/https access (from Internet)

arrmo, 2016-10-22:


tagyoureit, 2016-10-23:
I'm not even sure where to get started with this.  Need some guidance.

arrmo, 2016-10-23:
Yep, I figured I'd try to take care of this one - just wanted to make sure it was captured.

arrmo, 2016-10-29:
OK, a bit of digging here - and a stumbling block I think ... :(. I did get https redirect working (to http for now), using the information here: http://stackoverflow.com/questions/8541182/apache-redirect-to-another-port

And then, we can set up https pretty easily, info here (and matches to express API docs): http://stackoverflow.com/questions/11744975/enabling-https-on-express-js#comment41236410_11744975

BUT (there's always a but(t) ... ;)) - the socket.io traffic is all over port 3000, no matter what, right? So that kinda defeats the purpose of https, right? I admit, trying to allow access out of networks that only pass 80 or 443 ... but seems that may not be an option? Rather, just open another port on the firewall, and forget getting out of restrictive networks (given socket.io traffic)?

Thoughts?

arrmo, 2016-11-07:
First step with PR #41 - password authentication (disabled by default). Perhaps add to the documentation how to create the file (htpasswd)?

arrmo, 2016-11-08:
Addressed with PR #42 (and router port forwarding of course).

And for reference info (documentation?) - here is how I created the self-signed certificates, that work fine with Express,
http://www.akadia.com/services/ssh_test_certificate.html

Test it out, close if it works?

arrmo, 2016-11-11:
Closed - verified by @leftyfl1p (working for this individual as well). Provides http/https, and also authentication (if desired, configure in config.json).

closed 2016-11-11 by arrmo