tagyoureit/nodejs-Pentair #26: Bootstrap schedules wrap

tagyoureit, 2016-10-23:
Minor adjustment needed as the schedules are wrapping.

[image]

arrmo, 2016-10-23:
Agreed! Saw that as well in your pic, thanks for capturing this one. There is a (sort of?) way to customize it right now, but let's take care of bigger items first, then I'll explain how to use the one json file.

arrmo, 2016-10-29:
Does adjusting the min width in style.css address / correct this? Still trying to find a way to stop wrap, but this may fix it?

tagyoureit, 2016-10-29:
Yes.  Changing to 600 (from 570px) for me works great.  Maybe on shorter widths the days could appear as a second line under each schedule?  This would save space, as well.  Better for small screens like phones.

arrmo, 2016-10-29:
I like that! Do we do that by default? It may be cleaner. I'd still like this to auto-adjust (the panel width), but so far it's past my brain capacity ... :(.

arrmo, 2016-10-29:
Quick hack ... like this?
[image]

arrmo, 2016-10-29:
OK, border removed, padding ("combined" rows) minimized,
[image]

tagyoureit, 2016-11-02:
Love it!  Looks great.  :-)

arrmo, 2016-11-04:
Will push this sometime this weekend. Thanks!

arrmo, 2016-11-05:
Closed with latest PR#34 ... close this?

tagyoureit, 2016-11-05:
Yup, agreed.  I'll close it.

closed 2016-11-05 by tagyoureit