tagyoureit/nodejs-Pentair #29: Script to Output Data (for SNMP Tracking)

arrmo, 2016-10-29:
A script interface (node.js executable) to output key information, like temperatures, etc. => SNMP logging?

tagyoureit, 2016-10-29:
Are you thinking a separate executable from the current app, or another method internally to talk to SNMP?  Do you use SNMP currently?  This is probably another good place for standardizing.  We have socket.io (bootstrap); winston->socket.io (debug messages); winston->console; http->ISY.  

I actually think the best long term idea I have so far is Socket.IO --> external resource.  The bootstrap is already one external resource.  But we could move the ISY code to another app that accepts the Socket.IO connections and then spits out the REST URI's.  And it would basically be a template so that we could use any other "client" app to listen to the sockets and output to whatever they need.

arrmo, 2016-10-29:
Yep, agreed 100%. I use LibreNMS - it can be customized to grab the info over (extended SNMP), so then you could plot, for example, water temp over the year. Or whatever you want to track. Definitely a "later" thing ... :)

arrmo, 2016-11-11:
Addressed by PR #44 ... close this?

tagyoureit, 2016-11-29:
I took what you did in this PR and included it in the new Integrations framework.   Test it out there!  https://github.com/tagyoureit/nodejs-Pentair/wiki/Integrations-in-2.0

closed 2016-11-29 by tagyoureit

arrmo, 2016-11-29:
Need to look at this yet - the script I had was already doing what it needed to do ... :-). I was about to send it to the LibreNMS folks, but let me check it now given the latest changes.

Thanks!

arrmo, 2016-12-06:
OK, I admit - a bit confused by this. But for what I need, I can go back to my client.js script ... though I'm not sure it works fully now, given different messages being sent for different info. May need to request a send all.

Thanks!

tagyoureit, 2016-12-06:
Yes, the objects might be a little different.  You can just call them from the URL (http://localhost:3000/XYZ) where XYZ is all, pump, circuits, etc.   It should be pretty much the same as the client.js script otherwise.  I can look at it if you need help.

arrmo, 2016-12-07:
OK, need to play with this a bit. Just tried the URL -> much more output than the script generated (and more than SNMP wants). Will try to update the client script again.