tagyoureit/nodejs-Pentair #30: Add Configuration HTML File

arrmo, 2016-10-29:
Do we add a configuration "GUI" - HTML to allow configuration of panels ... or just go with the json file? Configs won't change much / ever, so not worth it?

tagyoureit, 2016-10-29:
I think the whole config.json file is getting a bit messy...  it includes equipment, net/socat, ISY, bootstrap.  If this is a high level effort, than probably skip for now.  Maybe come back and address this whole thing later.  

I'm not clear how the settings are being saved now.  Can you enlighten me?

arrmo, 2016-10-29:
If you look in the bootstrap directory, there is a json file - it's for enabling or hiding panels. The rest is stock info that's needed by the client.

arrmo, 2016-11-11:
Addressed by configClient.json - closing, but reopen if needed!

closed 2016-11-11 by arrmo