tagyoureit/nodejs-Pentair #35: More panel controls?

tagyoureit, 2016-11-06:
Now that we have so many panels, 7 (plus upcoming pumps=8), we could add some more fine grain controls.  Either hide panels altogether if they aren't used (for example, schedules with pumpOnly), and/or a way to collapse the panels (http://jsfiddle.net/Wc4xt/1052/).  This might tie in with the configuration HTML in the other issue.

arrmo, 2016-11-06:
FYI, this is already somewhat available in configClient.json (below),
	"panelState" : {
		"system": { "state": "visible" },
		"pool": { "state": "visible" },
		"spa": { "state": "visible" },
		"features": { "state": "visible" },
		"pump": { "state": "visible" },
		"schedule": { "state": "visible" },
		"eggtimer": { "state": "hidden" },
		"debug": { "state": "hidden" }
	},

Hide / make visible here!

arrmo, 2016-11-06:
OK, also check out PR #38 => Panel Collapse option.

Just one more thought (a bit more difficult, but ...) - also add a startup state for collapsed?

tagyoureit, 2016-11-07:
Sure... that would be good.  I think we're getting to the point of needing to be able to save these.  But that can wait.  If you want to enable manual defaults for now that would be awesome.

arrmo, 2016-11-07:
Already added the state ... :). Just edit clientConfig.json, set it as you want. It works!

tagyoureit, 2016-11-07:
Ummm, right.  You already said we can edit the config.  👍 

> also add a startup state for collapsed?

Sure... why not!

arrmo, 2016-11-07:
Sorry, what I meant is - I already set it up (with the latest updates) to also support startup as collapsed. Make sense?

tagyoureit, 2016-11-07:
Am I missing it?  In `clientConfig.json`, I only see
```
	"panelState" : {
		"system": { "state": "visible" },
		"pool": { "state": "visible" },
		"spa": { "state": "visible" },
		"features": { "state": "visible" },
		"pump": { "state": "visible" },
		"schedule": { "state": "visible" },
		"eggtimer": { "state": "hidden" },
		"debug": { "state": "hidden" }
	},
```

Visible/hidden.  But how do I set it up for collapsed/expanded?  Is that the same?  I thought `hidden` would remove it from view altogether?

arrmo, 2016-11-07:
It does! Sorry for the confusion. My local file is as below - but I didn't push this, was a bit gun-shy ... didn't want to break other folks setups (like the netconnect issue).
	"panelState" : {
		"system": { "state": "visible" },
		"pool": { "state": "visible" },
		"spa": { "state": "visible" },
		"features": { "state": "visible" },
		"pump": { "state": "hidden" },
		"schedule": { "state": "collapse" },
		"eggtimer": { "state": "hidden" },
		"debug": { "state": "collapse" }
	},

=> so use collapse, not hidden or visible.

tagyoureit, 2016-11-07:
Ah!  Ok, great... Tested and works PERFECTLY!  So closing this out...  Let's remember to add this to Bootstrap documentation (in README.md?) when you get to it.

closed 2016-11-07 by tagyoureit

arrmo, 2016-11-07:
Yep, agreed! For the documentation, do we just note the three options ... and really, hidden / visible is for what is shown, and under visible a further option for collapse (startup condition)?

BTW, this is really all we need to note, right? Perhaps I guess also ... don't change the names, they align to the HTML file.

Thanks!