tagyoureit/nodejs-Pentair #37: Update time formatting

tagyoureit, 2016-11-06:
Current time formatting coming from the controller/pumps is weak.  It currently might output something like 7:5 for 7:05.  Need to clean this up.

Possible impact on Bootstrap UI.  Right now, 12:05am shows up as 0:05.  Arrmo/tag will coordinate on this.

arrmo, 2016-11-06:
NP! Just to understand ... can you confirm the format the info is sent across in (for, say, 12 am)?

tagyoureit, 2016-11-12:
Right now, it's just a X:Y where X is the hours and Y is the minutes.  I think X is 0-23 and Y is 0-59.  Will try to get to this one soon.  Been focusing on a few other bigger items ATM.

arrmo, 2016-11-12:
No worries - will take care of it on the client side. Thanks!

arrmo, 2016-11-13:
OK, I can see where this is - fix is quite easy. Just one question though ... is this only on the schedules? Seems to be, just making sure.

For the current time, no formatting done on the client side, and I assume it's not needed ... right?

arrmo, 2016-11-13:
Pushed a fix - pull it in, check it there? Thanks!

tagyoureit, 2016-11-27:
One small item.  When it is exactly on the hour (x:00) it displays as x:000.

arrmo, 2016-11-27:
Will check on it! Is this for clock time, or schedule time?

Thanks.

tagyoureit, 2016-11-27:
I saw it in the clock time. Didn't check schedules.

arrmo, 2016-11-27:
Gotcha - thanks! Will dig into it. Right now I can't get any information (socket.io) from server to client - need to get through that first ... :(

arrmo, 2016-11-28:
Yep, it's there! Had to add the code below to catch it, and let it run until it hit the hour ... :).
		if (data.controllerTime.includes('00'))
			var test=1;

When this did hit, I checked - and data.controllerTime is "6:000 PM" ... so is this actually in index.js?

arrmo, 2016-11-28:
OK, now also seeing "6:202 PM".

tagyoureit, 2016-11-28:
I think I fixed it in the last push.  Check it out.

arrmo, 2016-11-29:
That works! BTW, why is Run Mode and Heater State not passed in 2.0-Alpha?
[image]

tagyoureit, 2016-11-29:
The heater status comes from the status packet, but because #32 (I'm not sure which one of the bytes correlates to the heater mode vs heater on) I didn't wire it up.  I can hook one of them up and hopefully myself or someone else can figure out exactly what it means.

Run mode... hmmm... I think this might have been something from the pump status early on.  Need to investigate.

Let's open up separate issues for this.  I think we can close the time issue for now.

closed 2016-11-29 by tagyoureit

arrmo, 2016-12-01:
Works for me - thanks!