tagyoureit/nodejs-Pentair #46: Chlorinator mode

midi30, 2016-11-14:
Hello,
In the present version of nodejs-Pentair1.0.3, it is not possible to use chlorinator without having at least one pump under control.
It would be interesting to setup a standalone chlorinator mode (without needing the pumps).
Thank you.

tagyoureit, 2016-11-14:
[Pentair_Pool-ZODIAC-Tri.pdf](https://github.com/tagyoureit/nodejs-Pentair/files/590407/Pentair_Pool-ZODIAC-Tri.pdf)

Thanks Midi.  I attached the doc you created so others can reference it.  

The workaround, as stated above for anyone finding this issue... if you want a standalone chlorinator, is to set `pumpOnly=1` for now.  Will fix in a future release to remove the dependency on the pumps running.

tagyoureit, 2016-11-26:
@midi30 I think I adjusted this fully with the 2.0 release.  Can you test?

@arrmo Can you add a chlorinator section to bootstrap?

arrmo, 2016-11-26:
I can, but I admit - don't have one here, so a bit more challenging to test (but not impossible!). A few questions, to try to get this right,
1) Can you send a sample JSON string ("data") for testing? Without a chlorinator, I don't have any input to use.
2) Do you want this as a separate panel?
3) What all to show in the panel (separate or with another)?

Thanks!

tagyoureit, 2016-11-26:
1.  {"saltPPM":2900,"outputPoolPercent":7,"outputSpaPercent":-1,"superChlorinate":0,"version":0,"name":"Intellichlor--40","status":"Unknown - Status code: 128"}

2.  Yes, separate panel would be ideal.

3.  Please show everything in the JSON except for version.  I don't know what that is at the moment.  
a)  Please show the Chlorinator status just like pool & spa.  The logic will be:  If either outputPoolPercent>0 OR outputSpaPercent>0 then the chlorinator will be on.  
b) One thing we will leave for later: The output Spa % is only valid with a controller.  Otherwise the pool % is the only one available.  But this would require a deeper understanding/exposure of the equipment config so we will leave it for now.  
c) To set the chlorinator level, the user should have a selector where they can select 0-101.  0 Would be off, 1-100 would be the chlorinator %, and 101 is super chlorinate (but it's not really important now).  But please verify the value is in this range.
d) when the chlorinator button/label is pressed, it should call `socket.emit('setchlorinatonr', level)`

arrmo, 2016-12-04:
Working on this, but to confirm - is the message in 1) above sent using socket.on('chlorinator')?

Thanks!

arrmo, 2016-12-04:
First code added, can someone check it out (need PR #52 included, it falls in this automatically). This is a fixed config for now ... need to do the following yet,
1) Add socket.on message (need to know what the flag is, as above)
2) Add input capability, and set level -> need a bit more description of what is desired, once looking over the code in place now.

Thanks!

tagyoureit, 2016-12-04:
```
        socket.on('setchlorinator', function(desiredChlorinatorOutput) {
            container.chlorinator.setChlorinatorLevel(parseInt(req.params.chlorinateLevel))
        })
```
This is what is on the server side.  You should create a basic input box that accepts a value of 0-101 and when the user clicks on the chlorinator button (whether it is green/On or blue/Off) it should send the `socket.emit('setchlorinator', value)`.  The backend will take care of re-emitting the correct status once it is set.  Is that what you mean by the flag???  And hopefully this answers both questions. 

It looks great!  See the pic.

[image]

arrmo, 2016-12-04:
Glad you like it! OK, step #1 ... PR #54 -> try it out, see if this works for you (to input the data). If it does, then I'll hook up the emit back, and we should be good on this one.

tagyoureit, 2016-12-06:
Works great!  Just pushed one small change from my side and it looks awesome.  👍

arrmo, 2016-12-07:
Excellent, and glad it helped to highlight an issue. Now ... on my side, enable socket.io input ... agreed?

Thanks!

arrmo, 2016-12-31:
Hi,

I never did hear back on this ... enable socket.io now? I admit, I don't have a Chlorinator, so I can't test it locally - but if someone can, I'll enable this now.

Thanks!

midi30, 2017-01-03:
@arrmo 
Hi,
I installed and run V2.0.4 with Zodiac Tri Chlorinator.
It is Ok for setting chlorinator values (called "set salt") including OFF and SuperChlorinator.
Some little things, as far as I could see
- Chlorinator status is not updated (and remain "STATUS") event when chlorinator send a new status value,
- Salt value is correctly maintained,
- How can we send Name & Version messages ?
- "Status Request" message is send every  5 seconds : is it not too much ?

Nevertheless, the present version is very usable 👍 
Thanks .... and happy New Year !!!

arrmo, 2017-01-07:
Yep, it won't update - I can enable that ... was waiting for someone who can test that it's working (i.e. updates are correct). Can you do that? If so I'll push the update this weekend.

Thanks!

midi30, 2017-01-07:
Yes, I can test it. No problem.

Thanks !

arrmo, 2017-01-08:
Updated! I just need tag to accept PR #57 - I believe I can do it myself, but I do prefer a second set of eyes on it (not approving my own changes .. :)).

Thanks!

midi30, 2017-01-09:
Thanks.
I did tests. 
As you can seen in the attached files, the only item updated is Salt PPM (from time to time with 4000, and from time to time to 2900).
I did tests with Intellicom = 0 or 1 OR Intellicouch = 0 or 1.
State, Name, Pool % and SuperChlorinate remain -1 (or false).
Is there something wrong ?
What can I do ?

[2.0.4+_09-01-2017_10h58.txt](https://github.com/tagyoureit/nodejs-Pentair/files/693182/2.0.4._09-01-2017_10h58.txt)
[2.0.4+_09-01-2017_11h03.txt](https://github.com/tagyoureit/nodejs-Pentair/files/693183/2.0.4._09-01-2017_11h03.txt)
[2.0.4+_09-01-2017_11h06.txt](https://github.com/tagyoureit/nodejs-Pentair/files/693184/2.0.4._09-01-2017_11h06.txt)
[image]
[image]
[image]
[image]

arrmo, 2017-01-09:
I'll let tag comment - as not sure this is a client side issue (but I could be wrong!).

I am glad to see you using the client side debug log ... :).

midi30, 2017-01-10:
Well ... What do you mean by "_client side debug log_" ?
The chlorinator is, in fact, far from here (1000km), so I cannot see what is on his screen, if that is what you mean.
What else can I do ?

arrmo, 2017-01-10:
Let's let tag comment - there may be some logs / information that will help here (in terms of messaging).

Thanks!

midi30, 2017-01-10:
Do you me to close, now ?

tagyoureit, 2017-01-10:
Hi, give me a few days to try this.  I'm slammed with a few other things right now.

midi30, 2017-01-11:
Yes of course.
Tell me if there is something I can do,

Thanks !

tagyoureit, 2017-01-22:
I'll reopen this on the new site: https://github.com/tagyoureit/nodejs-poolController/issues/12

closed 2017-01-22 by tagyoureit

arrmo, 2017-01-22:
I think the client side fixes are in place for this already - no? Just not sure what I need to do here, sorry!