tagyoureit/nodejs-Pentair #49: Add pump control to Bootstrap

tagyoureit, 2016-11-26:
@arrmo Can you add the pump control to Bootstrap?  You can see the initial iteration at https://github.com/tagyoureit/nodejs-Pentair/blob/master/public/pump.html.

arrmo, 2016-11-27:
Hi,

Are you thinking inside the Pump Panel (I assume)?

Thanks!

tagyoureit, 2016-11-27:
Ya... possibly.  But not everyone needs direct control over the pump.  Is it possible to do a "nested" container?  And that can be hidden in the config file for those that have controllers?  Or possibly just a separate one... I'm ok with either until we get some feedback from others that need the pump control.  Personally, I'll keep it hidden.

tagyoureit, 2017-01-22:
moved to https://github.com/tagyoureit/nodejs-poolController/issues/11

closed 2017-01-22 by tagyoureit