tagyoureit/nodejs-Pentair #52: Run mode on pool status

tagyoureit, 2016-11-29:
Is it necessary?  Research why it's there.

arrmo, 2016-12-01:
Or ... I could just add client logic = show as Running, if Pool or Spa are on?

tagyoureit, 2016-12-01:
I had this in the 1.0 code base:
https://github.com/tagyoureit/nodejs-Pentair/blob/master/index.js#L1030
`status.runmode = strRunMode[data[controllerStatusPacketFields.UOM] & 129]; // more here?`

I turned on my pool, but didn't see anything correlating to 129.  Let's remove it for now.  It would be redundant to having the pool/spa indicators on.  

If we wanted to make this a "summary" tab of everything that's on, then it might make more sense.  But I think it's ok for now just to remove it.

arrmo, 2016-12-02:
NP, will remove it! A bit behind on work at home right now, but will get to it this weekend.

Thanks!

arrmo, 2016-12-04:
Removed - PR #53 created.

closed 2016-12-17 by tagyoureit