tagyoureit/nodejs-Pentair #55: Lighting?

rayzorben, 2016-12-29:
Got my RS485 USB port, hooked it up to my RPI and got it working today, excellent job!

Is there any support for the lighting? What I am looking for is the ability to set the spa and pool lights to different colors.

If I use the 'Color Set' functionality I can do this through the Intellilogic software. There is also 'Intellibright' that allows me to set 1 of 5 colors, but it is set for both the pool and the spa to the same color.

The problem that I found out this holiday season is that although 'Intellibright' supports a Green and Red, 'Color Set' does not have a Red color. The bulbs can actually output a red, but I can't set the lights to Red and Green, so I am hoping to direct this manually.

Any ideas?

tagyoureit, 2016-12-29:
Hi,
Glad you got it working.  I have a lot of equipment, but unfortunately I don't have any of the LED color changing lights so I can't decode those packets locally.  

What you can try to do is go into the `config.json` and set `logConsoleNotDecoded: 1` and 'loglevel: debug`.  Then from your wired/wireless/screelogic controller toggle the colors and see what packets are sent.  Hopefully, there will be some packets that have RGB  values (eg [rgb example](http://www.rapidtables.com/web/color/RGB_Color.htm) with 0,0,0 = black all the way to 255,255,255 = white).  

See the [wiki](https://github.com/tagyoureit/nodejs-Pentair/wiki/Broadcast) for examples of how other packets are configured.

Let me know what you find.
Tag

rayzorben, 2016-12-29:
will try it out tomorrow and report back.

rayzorben, 2016-12-29:
Here's what I got. There are 2 light color modes. Intellibright where you can set a color specifically or a 'color shifting mode'.

There is also a 'Settings' panel where you can set the pool light separately from the spa light, and it communicates this back to the EasyTouch, but doesn't actually set these colors unless you hit 'Color Set' button.

If you want to give me an idea of the format, I can look into helping out with the project.

Set to Red

18:29:50.922 INFO Msg# 118 is NOT DECODED [165,1,16,34,96,2,195,0,1,253]
18:29:50.943 VERBOSE Msg# 119   Ack Message packet: 165,1,34,16,1,1,96,1,58
18:29:52.060 VERBOSE Msg# 121   is NOT DEFINED packet: 165,1,16,34,96,2,195,0,1,253

Set to Blue

18:30:51.924 VERBOSE Msg# 83   is NOT DEFINED packet: 165,1,16,34,96,2,193,0,1,251
18:30:51.928 INFO Msg# 83 is NOT DECODED [165,1,16,34,96,2,193,0,1,251]
18:30:51.944 VERBOSE Msg# 84   Ack Message packet: 165,1,34,16,1,1,96,1,58

Set to Green

18:31:53.331 VERBOSE Msg# 189   is NOT DEFINED packet: 165,1,16,34,96,2,194,0,1,252
18:31:53.337 INFO Msg# 189 is NOT DECODED [165,1,16,34,96,2,194,0,1,252]
18:31:53.352 VERBOSE Msg# 190   Ack Message packet: 165,1,34,16,1,1,96,1,58

Set to White

18:32:44.996 VERBOSE Msg# 331   is NOT DEFINED packet: 165,1,16,34,96,2,196,0,1,254
18:32:45.011 INFO Msg# 331 is NOT DECODED [165,1,16,34,96,2,196,0,1,254]
18:32:45.023 VERBOSE Msg# 332   Ack Message packet: 165,1,34,16,1,1,96,1,58

Set to Magenta

18:33:31.023 VERBOSE Msg# 432   is NOT DEFINED packet: 165,1,16,34,96,2,197,0,1,255
18:33:31.026 INFO Msg# 432 is NOT DECODED [165,1,16,34,96,2,197,0,1,255]
18:33:31.050 VERBOSE Msg# 433   Ack Message packet: 165,1,34,16,1,1,96,1,58

Caribbean

18:33:59.070 VERBOSE Msg# 498   is NOT DEFINED packet: 165,1,16,34,96,2,179,0,1,237
18:33:59.073 INFO Msg# 498 is NOT DECODED [165,1,16,34,96,2,179,0,1,237]
18:33:59.094 VERBOSE Msg# 499   Ack Message packet: 165,1,34,16,1,1,96,1,58

Party

18:34:43.352 VERBOSE Msg# 576   is NOT DEFINED packet: 165,1,16,34,96,2,177,0,1,235
18:34:43.355 INFO Msg# 576 is NOT DECODED [165,1,16,34,96,2,177,0,1,235]
18:34:43.375 VERBOSE Msg# 577   Ack Message packet: 165,1,34,16,1,1,96,1,58
18:34:44.485 VERBOSE Msg# 581   is NOT DEFINED packet: 165,1,16,34,96,2,177,0,1,235
18:34:44.488 INFO Msg# 581 is NOT DECODED [165,1,16,34,96,2,177,0,1,235]
18:34:44.508 VERBOSE Msg# 582   Ack Message packet: 165,1,34,16,1,1,96,1,58

Settings Spa Light to Cyan

20:51:12.847 VERBOSE Msg# 12156   is NOT DEFINED packet: 165,1,16,34,167,32,3,4,10,0,4,6,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,196
20:51:12.850 INFO Msg# 12156 is NOT DECODED [165,1,16,34,167,32,3,4,10,0,4,6,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,196]
20:51:12.891 VERBOSE Msg# 12157   Ack Message packet: 165,1,34,16,1,1,167,1,129

Settings Pool Light to Magenta

20:51:47.582 VERBOSE Msg# 12206   is NOT DEFINED packet: 165,1,16,34,167,32,3,12,10,0,4,6,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,204
20:51:47.584 INFO Msg# 12206 is NOT DECODED [165,1,16,34,167,32,3,12,10,0,4,6,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,204]
20:51:47.626 VERBOSE Msg# 12207   Ack Message packet: 165,1,34,16,1,1,167,1,129

Settings Pool Light to Blue

20:52:39.844 VERBOSE Msg# 12283   is NOT DEFINED packet: 165,1,16,34,167,32,3,8,10,0,4,6,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,200
20:52:39.847 INFO Msg# 12283 is NOT DECODED [165,1,16,34,167,32,3,8,10,0,4,6,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,200]
20:52:39.888 VERBOSE Msg# 12284   Ack Message packet: 165,1,34,16,1,1,167,1,129

Color Set (Used to Set the lights as set in the 'Settings [x] light to [color]' above.

18:35:47.517 VERBOSE Msg# 674   is NOT DEFINED packet: 165,1,16,34,96,2,160,0,1,218
18:35:47.520 INFO Msg# 674 is NOT DECODED [165,1,16,34,96,2,160,0,1,218]

Random

18:32:31.693 INFO Msg# 292 is NOT DECODED [165,0,16,96,255,1,8,2,29]
18:32:31.819 INFO Msg# 294 is NOT DECODED [165,0,16,96,255,1,8,2,29]
18:32:31.954 INFO Msg# 296 is NOT DECODED [165,0,16,96,255,1,8,2,29]
18:32:32.081 INFO Msg# 298 is NOT DECODED [165,0,16,96,255,1,8,2,29]

18:35:27.730 VERBOSE Msg# 652   Get ? packet: 165,1,16,34,231,1,0,1,192
18:35:27.785 VERBOSE Msg# 653   Set ? packet: 165,1,15,16,39,32,3,4,10,0,4,14,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,57

tagyoureit, 2016-12-31:
Thanks for posting these.  Been a busy few days.  Let me find some time to dig into this.

tagyoureit, 2017-01-19:
Hi, I started to look into this.

Couple of questions.
1.  Which of the following do you have: IntelliBrite® LED light, MagicStream® laminars, SAm® or SAL® color lights
2. It looks like the controller can accept color swim, color set, sync or smart start commands.  Do you know which one of these you are using?
3. Can you post your circuits from the following URL (http://computername:3000/circuit)  ?  Some of the lighting will map back to your circuits.
4. See page 46 here (http://www.pentairpool.com/pdfs/IntelliTouchOM.pdf).  There is also the configuration to assign lights (assuming Intellibrite, but also true for the other types) into groups.  

As you can probably tell, there are _many_ options with the pool lighting.  It's a tangled web.  :-)

jeffleus, 2017-01-19:
I actually don't know anything about the hardware as it was already there
when we bought the house and there were no manuals, etc.  The spa light is
just a basic incandescent light.  The pool light has a color wheel
mechanism in it that can run in two modes.  (1) Wheel fixed at the current
color.  (2) Wheel spins and changes colors from the light over time.  The
light is signaled to enter mode by toggling power ON-OFF-ON quickly.  Jenky
right?  I don't know if one of the color color wheel lights like SAm or SAl
match this feature set and have had trouble just researching online.
Probably need to hit a pool supply store.

No idea on the commands.  All of my hardware is older and very analog
driven w/ a binary state, except for the previously mentioned light.

I will post the circuit info after grabbing it tonight.  But I made some
progress already on my own.  I could see the circuits assigned to AUX1 and
AUX2, which are the relays connected to the spa light and pool light.  But
these did not show on the UI because I noticed you have a variable that is
set to hide the AUX circuits.  When I changed this, the AUX1, AUX2, and
AUX3 circuits appeared above the feature buttons... and they worked!  I was
able to figure out how to rename the circuits at the controller and after a
restart, the circuits appeared as named: SPA LIGHT, POOL LIGHT, and BLOWER.

I guess the part that has been the most confusing to me is in regards to
the "FEAUTRES".  I

jeffleus, 2017-01-19:
...Sorry I fat-fingered a pre-mature send...

Anyway, I was under the impression that I needed to be working w/ the
naming/assigning the features (esp since the UI was only showing feature
buttons).  But now, I think I am all set from the standpoint of your sfwr.

   - I can turn the pool on/off and change from NONE to HEATER mode and
   control set points.  I am cheap and don't heat the pool by default.
   - I can turn the spa on/off and change the setpoint for the heater.  I
   leave the spa in HEATER mode w/ a high setpoint (102) as this is usually
   the only way we use the spa.
   - I can toggle the SPA LIGHT, POOL LIGHT, and the BLOWER all from the
   buttons that now appear. (Since renaming the circuits, I was able to reset
   the hide AUX variable).
   - And, I can get the current system time, pool temp and air temp from
   the controller and properly reported.

The only thing that would be considered missing is the system reporting on
the heater status.  I just get a "--" output in the UI.  I imagine this
field in the system status should show when the heater is actively on and
heating the water?  Thoughts on where to look?  Do you think it might help
to have a default of show AUX, or maybe some notes in the docs?  I could
submit this as a specific issue and even submit a pull request to change.
Thoughts.

On the topic of automation and Alexa, I spent some more time looking into
options and the AWS IoT offering provides a much more comprehensively
secured solution for connecting Alexa and devices behind the home
firewall.  My next step is to get a RPi and get your software working
effectively.  In parallel, I plan to start playing around w/ the AWS IoT
service and see if I can mock something up.  This could then run as a
"device" on the RPi with an appropriate deviceShadow that controls/reports
all the state.  And it could repsond to the AWS IoT message that can be
configured to securely communicate w/Alexa or a mobile app for that matter.

-Jeff

tagyoureit, 2017-01-19:
Lots of good thoughts!

> I actually don't know anything about the hardware as it was already there
when we bought the house and there were no manuals, etc.

What controller do you have?  You should be able to see either by something on the outside/inside of the controller itself or by going into the menu (I think it is Diagnostics on the Intellitouch).

> The light is signaled to enter mode by toggling power ON-OFF-ON quickly.  Jenky
right?

Yes, very!  Do you have Intellibrite?  http://www.pentairpool.com/pdfs/IntelliBrite%205G%20OM.pdf  There are instructions there for power-cycling to get to different light modes.

> I could see the circuits assigned to AUX1 and
AUX2, which are the relays connected to the spa light and pool light.  But
these did not show on the UI because I noticed you have a variable that is
set to hide the AUX circuits.

Yes, where is this?  :-)  We can definitely note this.  Aux may in-deed mean that it is not assigned, but it is good to be able to see it as in your case.

> The only thing that would be considered missing is the system reporting on
the heater status.

Yes, this is still a little bit (pun intended) of a mystery.  I've tried a few times to nail down exactly what is happening but it has proved a little difficult.  Hence the 'help wanted' tag on https://github.com/tagyoureit/nodejs-Pentair/issues/32.    

> On the topic of automation and Alexa
Awesome!  Let's take this conversation to the Gitter forums and continue it there.

Glad you have made so much progress!

jeffleus, 2017-01-19:
> pdfs/IntelliBrite%205G%20OM.pdf There are instructions there for

I will check the controller version info tonight.  But EasyTouch-8 is the
retail model name.

Can't be Intellibrite since it is an incandescent bulb.

Just posted an issue on the github repo for the $hideAux variable with a
link to the line in code.

I will look at the #32 help wanted and do a little digging in my logs.

I will try and summarize the AWS IoT stuff on the gitter/home_automation
channel tonight.

tagyoureit, 2017-01-22:
moved to: https://github.com/tagyoureit/nodejs-poolController/issues/10

rayzorben, 2017-01-23:
@tagyoureit 

> Which of the following do you have: IntelliBrite® LED light, MagicStream® laminars, SAm® or SAL® color lights

I have IntelliBrite LED light

> It looks like the controller can accept color swim, color set, sync or smart start commands. Do you know which one of these you are using?

Any of the commands that say **Set** to **Color** are what I believe they call Smart Start commands. They set both lights to the same color. This is also part of the color mode series such as Royal, Caribbean, etc.. which changes the colors between a fixed set of colors.

Any of the commands that say **Settings** **Light** to **Color** are done from the Settings panel. These do not actually turn on the lights or set the color at that time. You HAVE to press "Color Set" to use these colors, and it is the only way to set the lights to different colors.

The 2nd to last one is the **Color Set** command.

I will post an answer to 3 and 4 hopefully soon. It has been raining to flood levels recently and I have to go outside and run the cable and leave the door open currently to test this out :P