tagyoureit/nodejs-Pentair #59: Detect (and Display) Freeze Protection

arrmo, 2017-01-06:
Ya, a bit of a cold spell here right now ... LOL.

It would be good to show when freeze protection is on / running.

rflemming, 2017-01-06:
I was actually planning on trying to figure that out today since it turned on last night. I've grabbed some logs and I'll look into it a bit later and update the wiki if I figure it out.

tagyoureit, 2017-01-06:
I looked into this the other week when my pool went into freeze mode.  It should be the value of 8 in what I am calling the Mode/Units/Etc (14) bit of the controller status (2) packet.  Check me and see if that's it.

rflemming, 2017-01-06:
It looks to me like that's correct.  I updated the wiki accordingly.  Actually the "Mode/Units/Etc" section I added earlier in the week since previously it only mentioned units.  Now we just need to figure out which bits are flipped when solar and heater modes are actively running.  I have solar and looked it during the summer, but couldn't nail it down.  I noted what changed, but more than one bit was being flipped so I'm not sure exactly which one means what yet.  I don't have a heater so I'll be of no use there.

tagyoureit, 2017-01-18:
I added this code in the new repo.  Will test it next time the temp drops to freezing.  Will re-open in new repo if this needs work.

@arrmo This is now an additional field returned from the temperature socket {freeze: 0} or {freeze: 1}.

arrmo, 2017-01-19:
Hmmm ... not seeing this in the temp socket - are you?

Thanks!

tagyoureit, 2017-01-19:
Yes, it's there.

`{"poolTemp":49,"spaTemp":49,"airTemp":46,"solarTemp":45,"freeze":0}`

Probably same issue as friendlyName.  Check to see you have the 3.0.1 code.

arrmo, 2017-01-19:
Ahhh ... I think I may see the issue. Do I need to move to the new repository to get the latest / fixes? I may be missing some of this - that could be the problem.

tagyoureit, 2017-01-19:
I had it in the 2.0.4 documentation, but since it's been quite a few weeks since I pushed new code I didn't test it there just now.  I may have "fixed" something in the meantime that didn't get pushed.  So yes, please do pull the latest from the new repo.

arrmo, 2017-01-19:
No worries. To understand ... move to the new repository then, right? And to do that ... fork from the new one, and push my latest updates to try to get in sync ... correct? Just want to do it right.

Thanks!

arrmo, 2017-01-19:
BTW, also track these tickets here - move to the new one as it makes sense?

tagyoureit, 2017-01-19:
> No worries. To understand ... move to the new repository then, right? And to do that ... fork from the new one, and push my latest updates to try to get in sync ... correct? Just want to do it right.

Yup, exactly!  Same process as before.  New location.  

> BTW, also track these tickets here - move to the new one as it makes sense?

Yes, we'll keep these open.  But any new ones let's move to the new repo.

arrmo, 2017-01-19:
OK, will fork to get a new repository set up - and then get my changes in place. Some PR's coming ... ;-)

arrmo, 2017-01-19:
Added to Pull Request (#1), over on new repository. Pull that in, and close this? Need to get cold weather to test it ... ;-).

One question though - I see the following in /circuits,
{"number":1,"numberStr":"circuit1","name":"SPA","circuitFunction":"Spa","status":0,"freeze":1,"friendlyName":"SPA"}

Is the freeze here just noting that freeze protection is enabled for this circuit, but the current state is what is in the temp message?

tagyoureit, 2017-01-20:
> Is the freeze here just noting that freeze protection is enabled for this circuit, but the current state is what is in the temp message?

Yes, exactly.  This just indicates if the circuit will run when the temperature is at the freezing point.

arrmo, 2017-01-20:
OK, then my code should work (and is in place) - just accept the PR that is on the new repository, and close this ... agreed?

arrmo, 2017-01-21:
PR included now, closing.

closed 2017-01-21 by arrmo