tagyoureit/nodejs-Pentair #63: Non-USB RS485 options?

gmorganVA, 2020-04-24:
Hi I see references in the documentation for this project to a USB to RS485 interface for connecting to the pentair equipment.  Have you had any success or can your recommend a more directly connected RS485 solution (SPI, TTL, etc)? I'm hoping to use a Raspberry Pi 3 as a network interface close to my pool automation (and a media converter for a fiber link back to my home network).  I want the optical isolation to prevent introducing noise / transients into my inside equipment.  If the USB interface is truly the best option I can come up with a physical arrangement to keep the USB connection reliable.

tagyoureit, 2020-04-24:
I'm not sure I fully understand your question.  The only way this app can connect to the pool equipment is RS485.  Someone smarter than me might be able to sync into the network communications (ScreenLogic on *Touch or direct comms on IntelliCenter).  

RS485 shouldn't introduce any noise and not sure what benefit the optical isolation would bring.

gmorganVA, 2020-04-25:
I need to read up more on RS485. If it is a differential interface then there shouldn't be a ground differential issues.  Given that pool automation equipment is outside and quite far from my wired home network I want to opto-isolate to prevent blowing out network gear if there are any transients.  I've already had a PoE switch burn up...fortunately I had an optical link in between it and the rest of my network.  Was looking for a RS485 interface "hat" for a raspberry pi that I would install inside the outdoor automation cabinet.  It'll get hot in there but not as hot as it would get in my attic. Maybe I'm just crazy...despite my EE degree I do a lot more software / sysadmin work.  Thank you for your time.

tagyoureit, 2020-04-25:
Let me know what you find. That’s all above my head. I know you can add resistors at the end of the line but that’s about it.   I’ll close this for now but feel free to add additional comments.

closed 2020-04-25 by tagyoureit

gw8674, 2020-05-03:
Something like this? 

> https://www.amazon.com/RS485-CAN-HAT-Long-Distance-Communication/dp/B07VMB1ZKH/ref=sr_1_3?dchild=1&keywords=RS485+CAN+HAT+for+Raspberry+Pi+4B%2F3B%2B%2F3B%2F2B%2FZero%2FZero+W%2FZero+WH%2CAllow+Long-Distance+Communication+via+RS485%2FCAN+Function+Onboard+CAN+Controller+MCP2515+via+SPI+Interface+Transceiver+SN65HVD230&qid=1588479638&sr=8-3

>