tagyoureit/nodejs-Pentair #64: Unable to open RS485 port

mzuniga51, 2021-09-30:
Hi.  First, thanks to you guys for this fenomenal contribution for us pool enthusiasts.   I have managed to install both the controller and dashboard in my Raspberry and I can access the web interface.  The interface says Ready but it is named Nixie Single Body instead of Intellicenter.  I got no information from the Intellicenter and I get the error found below.   Please let me know if there is anything I am doing wrong.  Is this a software error or can it be an error in my physical connection between my Raspberry Pi and my Intellicenter?  Many thanks.

[image]
!

tagyoureit, 2021-09-30:
You either don't have an rs485 connector or it isn't found at '/dev/ttyUSB0'. Did you see this [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details) page?

rstrouse, 2021-09-30:
Just for shucks turn off Bluetooth on the pi if you are certain that the rs485 adapter is installed properly.  If you have one that connects to the serial interface you need to disable the serial console in the pi configuration and it will be /dev/ttyS0