tagyoureit/nodejs-Pentair #66: NPM Start Error

gambler2841, 2022-07-13:
Hi,

I am getting errors while trying to connect to the pool equipment. I have had another Raspberri Pi connected previously,, but it died on me and now I'm starting over with a new Pi. It's been well over a year since I did it last time, so needless to say I'm rusty. Here is the error I'm getting when I run npm start&:

[image]

Any help would be greatly appreciated!

tagyoureit, 2022-07-13:
These are Typescript compile errors.  This might be one of three things...
1. You are not on the right version of Node.  Run `node -v` and see what version you are on.  v14 is ideal right now.
2. Did you install the node modules with `npm i` after you cloned the repo?
3. You may have some mismatched files somehow.

If you want to see a full set of directions for setting up the Pi you can follow [here](https://github.com/rstrouse/nodejs-poolController-dashPanel/wiki/nodejs-poolController-dashPanel-setup-and-configuration-on-a-Raspberry-Pi-for-the-novice) (there are some extra components installed but the main concept is solid).

You are safe to completely blow away the nodejs-poolController directory and follow steps 3-9 again.

gambler2841, 2022-07-14:
So I started over from steps 3-9 like you suggested. Here's what I got when I tried to install the correct versions of Node-Red and Node.js:

[image]

My current version of Node is v16.15.0. Should I downgrade to v14?

tagyoureit, 2022-07-14:
Yes node v14 is best. You also don't need node-red for this app. You can skip any steps related to that.